joshmarinacci/ElectronIDE #70: Electron Ace Editor accessibility

krperry, 2019-07-08:
Currently Electron is using Ace.  If you check Ace's issue tracker it has several accessibility problems that they are not able to fix.  The main one is blind users can not use the Ace edit field.   No screen reader works with it.

there are more blind users of Arduino than you might know and there would be more still in schools if the framerworks like this were accessible.  Note that Googles Go editor  which is an accessible editor like Ace could replace the Ace editor and is accessible.   

It would be nice if Electron would change from Ace to an accessible. editor.

wilsonmfg, 2019-07-08:
Electron hasn't been maintained for a long time, you should be using something else.

krperry, 2019-07-09:
This was an unhelpful comment.  First Electron is being used in other products.  Second you said use something else.  If you know of an or multiple accessible coding environment please list them.

  Note the Arduino IDE is only partially accessible.  It has 3 dialogs that need to be fixed.  The Arduino cli is accessible at the command line but does not give the quick easy access to boards these types of environments give.  Visual studio is accessible but you have to be able to use the inaccessible dialogs in Arduino IDE to make that work.  Eclipse is accessible but the extension is a c++ extension and does not use INO files.   So drive by answers are not helpful.

wilsonmfg, 2019-07-09:
#67

joshmarinacci, 2019-07-10:
I think you are confusing two projects. Electron was an Arduino IDE built on Node and Atom-Shell, part of the Atom project.  Around the time I stopped developing Electron the Atom-Shell project was renamed Electron, which makes it very confusing.  In any case, ElectronIDE, this Arduino tool, has not been developed for several years.  I stopped working on it because it became clear that advanced users didn't want a custom IDE, but would instead use their own tools; and Arduino Inc was building a web-based editor for the novice users. So ElectronIDE didn't have any reason to exist anymore.

krperry, 2019-07-10:
Yes I am confused.  Now I have to find out which electron is being used under the Let’s start coding IDE.

 

Ken

joshmarinacci, 2019-07-10:
You are probably looking for this:  https://electronjs.org/